osresearch/vst #8: Kits

stylesuxx, 2016-01-02:
Hey,
I saw the talk on 32C3 and was wondering if there are kits available? I just need this to put my oscilloscope to some good use :-)

On OSH the minimum quantity for the board is 3 pieces - which is too much for my personal use.

On the other hand I would be willing to order a medium run (30 pieces) on OSH, including the rest of the BOM and bundle them into kits - if I am allowed to and you are not planning on doing kits.

Price for the board would be: ~ 7€ / 7.5$
Price for the components, excluding Teensy: ~ 25€ / 27$
Totaling in at ~ 32€ / 34.5$ excluding the teensy.

Let me know what you think, also let me know if there is general interest in such a kit / group order.

quarterturn, 2016-01-03:
> Hey,
> I saw the talk on 32C3 and was wondering if there are kits available? I just need this to put my oscilloscope to some good use :-)
>
> On OSH the minimum quantity for the board is 3 pieces - which [...]

I don’t think the board design is finalized. In the meantime, just breadboard it. You can get something by just having the ADCs feed your scope if you tie AREF to 5v.

If you want an AC signal, you can add in the op-amps. I didn’t feel like buying the +/- 12v supply, but I did have an LT1054 on-hand and +/-5v regulators, so I ran the op-amps off +/-5v. Feeding you scope an AC signal should make it easier to get the display centered and use the voltage controls to fill the screen better.

I used an LM336Z-2.5 as the 2.5v reference, MCP33204P quad plus TL9721P dual rail-to-rail op-amps, and the LT1054. You can feed it 9v, let the LT1054 invert it, then regulate both to 5v using the usual 7805/7905 combo, or just let the LT1054 invert the 5v supply and live with the -4.7v you get. Since I was running on less voltage, I fed AREF on the ADCs using the 2.5v reference. With AREF at 5v, I saw clipping on the positive extremes, probably because the op-amps can’t reach the full positive rail. You should be able to get it all to fit on a standard solderless breadboard. 

|\ |  (¯  \/ |¯\  |V| |\ ¯|¯ |¯) | \/ | | | |¯\ (¯   /¯  /\ |V|
|-||_ (_  /\ |_/ @| | |-| |  | \ | /\ |^| | |_/ (_ . \_  \/ | |

stylesuxx, 2016-01-03:
@quarterturn That is some great info, thank you. Do you by any chance have a schematic you could share. Will source the parts and give it a try :-)

osresearch, 2016-01-03:
There is a schematic in our 32c3 presentation. I'll be posting the slides when I'm back from Germany. Until then, here is a rough example of how to breadboard it:

[image]

stylesuxx, 2016-01-03:
Alright, I am coming  near to my final part list.
Since I will not be using W & Z I only need one MCP4922 - but will still order 2 just in case I can get hold of some display supporting it.

* 1 x Teensy 3.2
* 1 x RB-0512D (+- power converter)
* 2 x MCP4922 (12 Bit DAC)
* 1 x 7805 (+5V)
* 1 x 7905 (-5V)
* 1 x LM336-Z2,5 (2.5V reference for MCP4922)
* 1 x AD822 (rail to rail Op Amp)

I could not find TL972IP on reichelt so I browsed through their available rail to rail op Amps, and found the [AD822](http://cdn-reichelt.de/documents/datenblatt/A200/AD822%23AD.pdf) which should be OK I guess.

The only thing I am not sure about yet:
The DAC outputs voltages between 0..2.5V
The Op Amp has a range of -5..5V (10V)
This means on the Op Amp I need a gain of 4 but need to offset the DAC's output by -1.25V. I thought about doing this with an adder like so:
[image]

Does this make sense, or is there a better way to do it?

quarterturn, 2016-01-04:
I posted the schematic here for you as a graphic file: http://imgur.com/efAkezt

Before ordering parts, it may be best if you don’t have a lot of experience to order what Trammel used exactly on the schematic, so you can copy it. For example, I used the LM336-Z2,5, but it’s a shunt regulator and works differently from MAX6102; it needs a 2.2k resistor connected V+ and then it is connected to ground. The 2.5v appears between the regulator and the resistor.

I used rail-to-rail op-amps as I had them on hand, but they aren’t necessary if you provide something more than +/-5v. Just don’t use something super-slow like LM324. You need 1 quad op-amp and 1 dual op-amp. There is a 5v reference on the schematic connected to AREF on the DACs, but I just used the 2.5v instead. It cuts down on the voltage swing out of the DACs, but it’s not a problem if you are using an oscilloscope.

You don’t need anything other than the Teensy 3 and the two DACs to see an image. The other parts make it easier to get a good image on a scope and are only necessary to drive a vectrex display, which expects certain voltages.

Good luck!

> Alright, I am coming near to my final part list.
> Since I will not be using W & Z I only need one MCP4922 - but will still order 2 just in case I can get hold of some display supporting it.
>
> 1 x [...]

|\ |  (¯  \/ |¯\  |V| |\ ¯|¯ |¯) | \/ | | | |¯\ (¯   /¯  /\ |V|
|-||_ (_  /\ |_/ @| | |-| |  | \ | /\ |^| | |_/ (_ . \_  \/ | |

osresearch, 2016-01-04:
If you're using an oscilloscope or a vectorscope with position adjustments, you probably don't even need the voltage ref, opamps and dual supply.  You can just connect the DAC output directly to the X+ and Y+, and ground the X- and Y-.

stylesuxx, 2016-01-04:
@quarterturn thanks for the schematic - this helps a lot.
@osresearch Oh, yes - this makes sense. Since I most probably will never treat myself to a Vectrex, I think this will be the way I'll build it.

Thanks for all the input, I'll keep you posted.

stylesuxx, 2016-01-11:
Hey guys, just wanted to update you. My Teensy arrived yesterday so I hooked it up to the DAC and checked what I could get out of it without any Op Amps and -12V/+12V power supply - wasn't that great but working quite OK.  
So today I added -12V/+12V power supply, 2.5V reference and one OpAmp for X and Y each, so the voltage is redistributed from 0 - 2.5V from the DAC to -1.25 - 1.25V than another Op Amp after that to amplify  X and Y. Works like a charm now. Really crisp edges, some time this week I will check out how it works with Mame. Having a lot of fun with it, so thanks again for the hard work and great input.

So my final BOM was like so:
* 1 x Teensy 3.2
* 1 x MCP4922
* 1 x RB-0512D
* 2 x AD822 (since I could not find a quad Op Amp)
* 1 x LM336-2,5

Leesam, 2016-01-20:
I finally got this together on a breadboard, i'd like to try it on my Vectrex. For now I've got it connected to an older 20MHz Dual trace oscilloscope (in XY mode). It's not drawing too accurately and I'm not sure if it's because of the scope or if I've got something not filtered/setup properly in my circuit. Any ideas where i should start?

https://goo.gl/photos/gMTSYZRZmBPqCK4t9

osresearch, 2016-01-20:
[image]

We have almost the same scope...  I see very similar trails, although not quiet as large, when using it with the board. The axes don't seem to be as fast as I would expect a 20 Mhz scope to be, which makes me wonder if it is an impedence mismatch or capacitance in the leads.

Leesam, 2016-01-20:
Here's what i'm getting on the Vectrex: https://goo.gl/photos/BTXxVh3ppR9JR8Br6 much better image than the scope but i've got an offset.

osresearch, 2016-01-20:
What are you using for the center reference voltage? A second DAC or a voltage divider? 

You should turn down the brightness to help prevent burn in, especially on the center of the screen.  Looks like X and Y are swapped as well.

Leesam, 2016-01-20:
I had the brightness turned up a bit to get a clear pic, but i keep it pretty low to prevent burn in. Just unswapped X and Y, so the picture is right side up now. 

I'm using a 2.5V reference REF192GPZ into IC9A/9D. It's putting out  about 2.487V. I'm not buffering it with an opamp as shown in the schematic.

Leesam, 2016-01-21:
I've got a clean test pattern now. Going by the schematic here: http://imgur.com/efAkezt it asked for vref2.5 into the opamp. 

I buffered the vref5 and used it as the Vdd and Vref on both DACs because they can only swing as high as VDD. With USB power i was getting something like 4.3V which was lower than the vref5 i had. 

I also used the vref5 into the opamps as the center reference and it pulled the image down into place perfectly.

Also, following the schematic i tried to pull Z from pin 10 of IC2, that was the DACW/vref channel, when i swapped it to pin 14 on IC2 the brightness started working.

Looks a bit bright in the image but that's just my cell camera overexposing: https://goo.gl/photos/ZjRVPvebWfdWR5mD8 (still looks pretty meh on the scope, but it's the vectrex i've got it configured for)

The only issue i've got left is when the processingDemo runs the vectrex dims at some points in the animations. At some points i lose the image completely, SwarmDemo basically wont show anything.

quarterturn, 2017-09-13:
I decided to permanently solder a version of the most current v.st board schematic. I used an Adafruit perma-proto board, which is the same size as the breadboard. The whole thing, dc-dc, voltage refs, and a dc power jack included, all fits.

Anyhow, I find I can not use my fast, modern MC33204 quad r2r op-amps. I'm guessing they oscillate, as total power draw goes to 630 mA at 5V and there's no display. I had to fall back to the old TL084 to get things working. Same deal on the buffer for the vrefs, had to use an old LM1458n there too. I'm not an analog person so I'll have to see what needs to go in the feedback loop to quell the osciilation.

I wonder if the RB-0512D is adequate. Total current draw when displaying the static teensysv.ino test screen is around 240 mA, and the converter is running at around 135 F. If you run it long enough, the converter apparently overheats and goes into shutdown mode, so I've got to fix it. The rest of the chips are fine - nothing more than 90 F anywhere. The RB-0512D is rated for about 42 mA per channel; I'd tested on the breadboard using an LT1054 which can do 100 mA (and also cheated a bit by running the whole analog side at +/- 5V, with an LM7805 handling the +5 rail.

quarterturn, 2017-09-13:
I added 8 pF across the feedback resistors on the TL084. It definitely sharpens up the display. I can now read the small text in the right quadrants. It definitely helps to un-fuzz the display. I also noticed I'd left off the 0.1 uF cap between GND and AGND. Adding that sharpened up things ever further. I can now zoom in and see the steps in the DAC output.

joyandrob, 2019-10-04:
> If you're using an oscilloscope or a vectorscope with position adjustments, you probably don't even need the voltage ref, opamps and dual supply. You can just connect the DAC output directly to the X+ and Y+, and ground the X- and Y-.

If I go this route (just to get started) do I not provide any additional voltage to the DACs? ...IOW there would be no +5v connection?

quarterturn, 2019-10-04:
> If I go this route (just to get started) do I not provide any additional voltage to the DACs? ...IOW there would be no +5v connection?

Give the VREF inputs on the DACs the +5V, then the output will swing from 0 to 5V.

joyandrob, 2019-10-04:
> Give the VREF inputs on the DACs the +5V, then the output will swing from 0 to 5V.

So, if the Teensy 3.2 only outputs 3.3V then I'm going to have to provide a separate 5V source to the DACs?

joyandrob, 2019-10-05:
> So, if the Teensy 3.2 only outputs 3.3V then I'm going to have to provide a separate 5V source to the DACs?

Found my answer here: http://little-scale.blogspot.com/2016/11/teensy-and-mcp4922-dual-channel-12-bit.html  It appears that the 4922's can run under 3.3V, so I can power them from the Teensy 3.2.

Got Trammell's vector mame compiled and running on my macbook today.  Had to find an older version of SDL2 (2.5) to get it to compile correctly.  

Can't wait to get my Oscilloscope and Teensy in the mail and start playing : ^ )

Thanks for the reply quarterturn!  ...i was afraid that this thread might be way dead after 2 years.

joyandrob, 2019-10-09:
All wired up on the breadboard.  I had to change the time.h include statement to the new timelib.h in order to get the Teenysv firmware to compile.  Also, I had to change the setup() size() statement to use P3D calls as the "processing" demo was throwing errors concerning the use of modelX() and a 2d library.  While waiting for my oscilloscope to show up in the mail I got bored and wanted to get a feel for the voltages being output by the controller board. ...to see if I've wired this thing correctly. I could see that the X and Y voltages were changing via my multimeter, but I couldn't tell if they were correlated to the vector positions being displayed. After some thought I figured that I might be able to "hear" the change in waveform since the voltages and current are close to what you would find on headphones. I wired up some cheap headphones to the X and Y output voltages and bam... I was definitely able to hear audible differences in pitch when vectors were draw in different parts of the screen. So, the circuit and software must be close.  Hope these notes are ok to leave here...just looking to pay it forward for anyone that might try this in the future.