osresearch/vst #19: Options Setting Brightness

jacobjoaquin, 2016-01-16:
I think we should consider how brightness is handled in the Vst methods. The current approach is handles brightness by passing a boolean as the first parameter, as demonstrated in the following signatures:

```java
// class Vst
void line(boolean bright, float x0, float y0, float x1, float y1)
void line(boolean bright, PVector p0, PVector p1)
void rect(boolean bright, float x, float y, float w, float h)
void point(int bright, PVector v)
```

Here is a list of potential options (more suggestions welcome):

1. Keep the methods as they are.
2. Re-arrange the order of parameters, placing brightness as the end, and then when no 5th parameter, it defaults to normal brightness. For example:

```java
// class Vst
void line(PVector p0, PVector p1)  // Normal brightness
void line(PVector p0, PVector p1, true)  // Bright
void line(PVector p0, PVector p1, false)  // Bright
```

3. Map the `stroke()` value.
    - While this works, the mapping of 256 points to 3 (off, normal, bright), seems off.
    - Need to account for overall brightness, since stroke is a combination of RGBA.
    - Need to account for `colorMode()` as well. It's possible to change the range from 0-255 to some other arbitrary range.
    - If there are plans to up the resolution of the z-index, then `stroke()` makes much more sense.
    - Here's how it would look:

```java
stroke(128);
vst.line(p0, p1);  // Normal
pushStyle();
stroke(255);
vst.line(p2, p3);  // Bright
popStyle();
vst.line(p4, p5);  // Normal
```

4. Create a new function that works like `stroke()`, but specifically for brightness.
    - Could respond to `pushStyle()` and `popStyle()`.
    - This 
    - Possibilities:

```java
// class Vst
void bright(float b)  // 0 - 255
void bright(int b)  // 0 - 3
```

5. A combination of the above. Though the trick here is trying not to over complicate it by trying to account for possible use.

Personally, I think I'm in favor of 2 and 5. Thoughts?

jacobjoaquin, 2016-01-18:
I've recently implemented a limited version `beginShape()`, `vertex()`, and `endShape()`. With these, I feel we're beginning to hit a limit with adding a `boolean bright` parameter. Here are the current signatures for the shapes:

```java
void beginShape()
void vertex(boolean bright, PVector p)
void vertex(boolean bright, float x, float y)
void vertex(boolean bright, float x, float y, float z)
void endShape()
void endShape(boolean bright, int mode)
```

Adding `boolean bright` is adding unneeded clutter, and I think there is a more traditional *Processing Way* of dealing with it. I'm now leaning more towards 3 and 4 in the above post. I'm willing to try implementing various versions of handling brightness to get a good read on the advantages and disadvantages of each.

osresearch, 2016-02-06:
This is no longer relevant, right? We are setting the brightness from the `stroke()` value and using the 4-byte command protocol to set each output line, which should allow us to remove all of the `boolean bright` arguments.

closed 2016-02-06 by osresearch

jacobjoaquin, 2016-02-06:
Correct. We're setting brightness with `stroke()`. One thing that was added in since we're using stroke as brightness is that `vst.colorStroke` was created to allow users to set preview stroke color. It defaults to black, but perhaps we should change that to white, assuming users will naturally opt for a black / dark background. Would be one less thing for users to have to set by default.

Here's an example from BrickBreaking101:

```java
  vst = new Vst(this, createSerial());
  vst.colorStroke = color(220, 220, 255);
  blendMode(ADD);
```