osresearch/vst #22: Teensy 3.2 SS chip select problem

callimero, 2016-03-31:
Hi,
I tried to get it running on a Teensy3.2 wich has as I understand a slightly different port setup on the SPI Ports?

The sketch compiles and uploads, I have a correct CS/SS signal on Pin 6, on PIN 10 however is NOTHING. My DACs working just fine, If I connect the second to PIN6 I have the same analog out on both. 

I tried changing this part but no joy.
	// configure the output on pin 10 for !SS0 from the SPI hardware
	// and pin 6 for !SS1.
	CORE_PIN6_CONFIG  = PORT_PCR_DSE | PORT_PCR_MUX(2);
	CORE_PIN10_CONFIG  = PORT_PCR_DSE | PORT_PCR_MUX(2);

Help!

osresearch, 2016-03-31:
The code should work with teensy 3.0, 3.1 and 3.2. Do you have `CONFIG_VECTORSCOPE` or `CONFIG_VECTREX` defined? In vectorscope mode the second DAC is not used since there is no brightness control.

callimero, 2016-03-31:
Hi Trammell,

seen you presentation on CCC, great things.

Now while we talk.... I use TWO mcp4921... NOT a mcp4922.... Grrrr.... Stupid me! Not sure if I can hack the source to use two mcp4921 for x/y.

Sorry for the Fuzz.
Carsten

osresearch, 2016-03-31:
The code can be hacked to do that, although you won't have any brightness control...
The code in `goto_x()` and `goto_y()` can modify the CS pin to be used for each point. The DAC channel will have to be set to 0 since the mcp4921 has only one channel.

https://github.com/osresearch/vst/blob/master/teensyv/teensyv.ino#L589

callimero, 2016-03-31:
Hmm. I tried to poke around in mpc4921_write() (the naming 4921 was also part of my confusion)

It at least produces something quadratic. Noisy thingy... http://www.pasteall.org/pic/101485

```
static void
mpc4921_write(
	int channel,
	uint16_t value
)
{
	value &= 0x0FFF; // mask out just the 12 bits of data

if (channel==0)
{
      digitalWriteFast(SS3_PIN,LOW);
      digitalWriteFast(SS_PIN,HIGH);

}
else
{
      digitalWriteFast(SS3_PIN,HIGH);
      digitalWriteFast(SS_PIN,LOW);
}

 value |= 0x7000 | (0 == 1 ? 0x8000 : 0x0000);

#ifdef SLOW_SPI
	SPI.transfer((value >> 8) & 0xFF);
	SPI.transfer((value >> 0) & 0xFF);
#else
	if (spi_dma_tx_append(value) == 0)
		return;

	// wait for the previous line to finish
	while(!spi_dma_tx_complete())
		;

	// now send this line, which swaps buffers
	spi_dma_tx();
#endif
    digitalWriteFast(SS3_PIN,LOW);
    digitalWriteFast(SS_PIN,LOW);  
}
```

callimero, 2016-03-31:
Thanks for your help, thanks to a shipping flatrate I just ordered a mpc4922, so in a few days I will be on ball park. That chip is even cheaper than two 4021.... grmblll.

But these 4921 will come in handy when I added the Z-Axis to my scope (hameg 205-2 it is internally prepared to that, just ON/Off I guess). Or maybe I use the also prepared TTL Z-Axis.

callimero, 2016-03-31:
Whhaaaa, I could not stop poking.

I had to understand the spi_dma_cs global var, now it works. Not all lines are going back to where there should I guess. Not sure if it is my hack or my scope? 

[image]