osresearch/vst #24: No resistor values for R16, R17, R20

crdotson, 2020-04-05:
The pcb/teensyv.brd file doesn't have resistor values for R16, R17, R20 that I can see.  If someone can give me a hint on what they should be (10k?), I can try a PR for an updated board file, and probably add digikey part numbers for easier ordering.

quarterturn, 2020-04-06:
Not sure why it's needed at all, since VREF is already current-limited at the first op-amp stage by the 10K resistor there, and as well the first op-amp should remove the 2.5v offset. Try it without.

Take a look here https://www.nutsvolts.com/magazine/article/the-arduino-graphics-interface-part-1, I think this is a better design in that it uses a sample-and-hold on the outputs to synchronize them and reduce glitching. Also the Due has better DMA performance (though a 3.5 or 3.6 might beat it with raw CPU power).

crdotson, 2020-04-10:
Thank you!  I actually already have a Due with AGI loaded but decided to build this one because of the existing MAME patches.

Swap-File, 2020-09-24:
In the docs I noticed that it says "TC33X trimpot (20K, allows gain up to 2X)"

This means R16 & R17 should be 10K to give you a max gain of 2x. 

R20 I'm less sure of, I'm still investigating, but it shouldn't matter much.  Maybe 75 ohms like the other outputs?

english1234, 2020-12-11:
I built a working VST several years ago, and looking at my notes, I used 10K for R16 and R17, and 100 ohms for R20 (I had noted that it should have been 75).